wayoda/LedControl #4: Minor test issues in setScanLimit() and setIntensity() methods

bertsierra, 2015-01-26:
I’ve spotted a couple of minor issues with the setScanLimit() and setIntensity() methods.  The test ‘limit&gt;=0 || limit&lt;8’ on line 88 is always true, so spiTransfer is called even with bad limit values.  Line 88 should be corrected as shown below:

<code>
    if(limit&gt;=0 && limit&lt;8)
</code>

Similarly, on line 95 there is a test ‘intensity&gt;=0 || intensity&lt;16’ which is always true no matter what.  Line 95 should be corrected as shown below:

<code>
    if(intensity&gt;=0 && intensity&lt;16)
</code>

wayoda, 2015-01-26:
Hi bertsierra, 
thanks for reporting the issue. I pushed the change to the repo, but before I do a new release I have to a hardware test (which will not happen before tomorrow). 

Cheers Eberhard

wayoda, 2015-01-27:
Tested your fix on my hardware and made a new release!

Thanks @bertsierra

closed 2015-01-27 by wayoda