wayoda/LedControl #5: image is mirrored on matrix

sebulica, 2015-04-25:
i've tested your library on a max7219cgn controlled matrix 8x8.the "images" were mirrored on the matrix.for example "d" was "b".i've modified the library a little and now works like a charm :)

i've modified the spiTransfer from LedControl,bot setRow and setLed

i've added "7-row+1" to make it look normal

wayoda, 2015-04-27:
This might have to do with a different wiring or even different mental conceptions about where column 0 is on a matrix. 

Anyway, I wrote the library 4 years ago on nobody complained about this. So I'm not going to change the library code, because that would break all the code for **every** user of the library who has to recompile a project.

But thanks for the hint and I'm glad the code was useful for you.  
Cheers 
Eberhard

closed 2015-04-27 by wayoda

marcelstoer, 2015-09-06:
Same here...but I guess it really boils down to the question "where is led 0"? If I run the sample sketch which displays "Arduino" I get the correct orientation if the wires of the board point upwards, the characters are mirrored, though. Below image shows the start of the `single()` function which lights up every led one by one -> it started top _right_:

[image]

tibfox, 2016-08-11:
@sebulica thx for your solution. had the same problem ;) works like a charme !

of course big thanks @wayoda ;)