wayoda/LedControl #6: printNumber(int v) function in the examples

LargoLobo, 2015-08-11:
Must assign "false" to the boolean variable "negative" up front. If not, it defaults to "true" and all the displays will be negative. Also, don't need to multiply v=v* -1;  It doesn't matter, since you are building the answer one digit at a time separately, and the sign digit is assigned regardless of this statement.

wayoda, 2015-08-12:
> Must assign "false" to the boolean variable "negative" up front. If not, it defaults to "true" and all the displays will be negative.

Fixed. C(++) is not my hometurf. Thanks for pointing this out.

> Also, don't need to multiply v=v* -1; It doesn't matter, since you are building the answer one digit at a time separately, and the sign digit is assigned regardless of this statement.

True but I left it in the demo code. I expect some readers have little programming experience so I try to make the code very explicit.

wayoda, 2015-08-28:
@LargoLobo If you don't have any objections, I close this issue now?

Feel free to reopen it if you think the demo code needs to be improved in any other way.
Thanks Eberhard

closed 2015-08-28 by wayoda