wayoda/LedControl #7: Bug in function setLed()

Darkhack01, 2015-08-25:
You can set only one LED in the same row. If you set another LED in the same row the other turns off!

wayoda, 2015-08-28:
Sorry, but can't confirm this behaviour. Here is stripped down version of the LedControl matrix example that blinks/lights up all of the leds on the matrix one after the other. Do you see a different result when you run this code with your hardware?

```
//We always have to include the library
#include "LedControl.h"

/*
 Now we need a LedControl to work with.
 ***** These pin numbers will probably not work with your hardware *****
 pin 12 is connected to the DataIn 
 pin 11 is connected to the CLK 
 pin 10 is connected to LOAD 
 We have only a single MAX72XX.
 */
LedControl lc=LedControl(12,11,10,1);

/* we always wait a bit between updates of the display */
unsigned long delaytime=75;

void setup() {
  /*
   The MAX72XX is in power-saving mode on startup,
   we have to do a wakeup call
   */
  lc.shutdown(0,false);
  /* Set the brightness to a medium values */
  lc.setIntensity(0,8);
  /* and clear the display */
  lc.clearDisplay(0);
}

/* 
 This function will light up every Led on the matrix.
 The led will blink along with the row-number.
 row number 4 (index==3) will blink 4 times etc.
 */
void single() {
  for(int row=0;row<8;row++) {
    for(int col=0;col<8;col++) {
      delay(delaytime);
      lc.setLed(0,row,col,true);
      delay(delaytime);
      for(int i=0;i<col;i++) {
        lc.setLed(0,row,col,false);
        delay(delaytime);
        lc.setLed(0,row,col,true);
        delay(delaytime);
      }
    }
  }
}

void loop() { 
  single();
  lc.clearDisplay(0);
  
}
```

wayoda, 2015-08-29:
No update on the issue from the OP. 
Closing as "invalid"

closed 2015-08-29 by wayoda