wayoda/LedControl #8: Performance improvements with use of SPI API

pasquyonline, 2015-09-20:
Hi, I have changed just 2 rows of the LedControl.cpp so that the library can use the SPI protocol.
I added 
1. <code>#include &lt;SPI.h&gt; </code> as the first row, 
2. <code>SPI.transfer(spidata[i-1]);</code> at row 206 instead of <code>shiftOut(SPI_MOSI,SPI_CLK,MSBFIRST,spidata[i-1]);</code>

With this simple changes now I'm able to show a message with an high scrolling rate (depending also on how you initialize the SPI interface). 
This is the initialization chunk in the <code>setup()</code> function:

<code>...</code>
<code>SPISettings settingsA(8000000, MSBFIRST, SPI_MODE0); </code>

<code>void setup() {</code>
<code>  // initialize SPI:</code>
<code>  SPI.begin(); </code>
<code>  SPI.beginTransaction(settingsA);</code>
<code>  ...</code>
<code>}</code>
 
[image]
[image]

wayoda, 2015-09-20:
Yes, thanks for the enhancement. 

I know that hardware SPI is *much* faster than my bit-toggle solution. By the time I wrote the LedControl library the arduino IDE did not have a SPI-library and using SPI would frequently clash with  other SPI-controlled devices. 

I never changed the LedControl implementation because (for me) one of the main features is that you can **use any 3 IO-Pins** to connect your device.  You are not tied to the Hardware-SPI pins.

If you don't mind I'll leave this issue open so others can see how you did it...

Thanks for the contribution
Eberhard

pasquyonline, 2015-09-20:
OK, thanks for your answer. I'm agree to leave the issue open.
If you are interested, I have also do a porting to the **Raspberry Pi** (very easy thanks to your clean code, always with the SPI) that use the **bcm2835** library.

nkolban, 2015-12-28:
I love the library but would also like to have the option of hardware SPI.  Can we not merge the two techniques and give the users the option of hardware vs bit banged?

gordoste, 2019-05-03:
I have a created fork that allows the choice between hardware and software SPI.