wayoda/LedControl #10: Doesnt work with esp8266 based boards

richms, 2016-01-01:
In file included from C:\Users\richard\Documents\Arduino\libraries\LedControl\src\LedControl.cpp:28:0:
C:\Users\richard\Documents\Arduino\libraries\LedControl\src\LedControl.h:30:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
Error compiling.

sej7278, 2016-01-08:
remove this line, then it works great!

https://github.com/wayoda/LedControl/blob/master/src/LedControl.h#L30

pcvm, 2016-02-12:
To save everyone looking this up, would it possible for this very useful library to have the following context diff applied?  Thanks.

[LedControl_context_diff.txt](https://github.com/wayoda/LedControl/files/127557/LedControl_context_diff.txt)

sej7278, 2016-02-12:
actually this is the proper fix, as it still lets pgmspace work with the esp8266 for 7-segment displays (its not needed for led matrices):

https://github.com/sej7278/LedControl/commit/d82d522791b805ebaebe926c45ca18a1ba35a3cc

MStylesMS, 2016-07-30:
Using this for a 7-segment display this worked fine.  Thanks!

1n5aN1aC, 2016-11-15:
I second this!  @sej7278 fix worked great!

ianbren, 2017-03-29:
#if (defined(__AVR__))
#include <avr\pgmspace.h>
#else
#include <pgmspace.h>
#endif

endriu91, 2018-02-16:
@wayoda don't You have time to repair this ? :(
This is great [https://github.com/wayoda/LedControl/issues/10#issuecomment-290219444](url)