wayoda/LedControl #15: more the 8 daisy chained

klarson18, 2016-03-25:
I would like to "daisychain" 10 7219 chips using ledcontrol.h but I am only successful with 8.  Is this a software issue or a chip issue?  Thanks

voneiden, 2016-04-15:
As far as I know the packet size for the chip is 16 bytes, which has 8 bytes of data. That means a maximum of 7 no-op bytes and as such it is limited to a maximum of 8 chips.

https://datasheets.maximintegrated.com/en/ds/MAX7219-MAX7221.pdf

klarson18, 2016-04-15:
Thank You for the response!  Makes sense.
Kyle Larson

lsxiao, 2017-10-21:
[image]

voneiden, 2017-11-03:
Yeah, I remember this was a wrong conclusion. More than 8 can be chained.

akhileshghanshyam, 2018-02-08:
@klarson18 can u please send me the code to run the 8 matrix chain. because i have problem with my code     
i have tried my best to make it run 8 matrix chain instead of 4. it will be very kind of u. my email id ghanshyam6298@gmail.com

akhileshghanshyam, 2018-02-08:
@voneiden can u please help me to chain 10 (8x8) matrices. I have code which is capable of running 4 matrices. i am trying to make it to run 10 matrices. but no succeeded yet. can you help me in any way. My email id ghanshyam6298@gmail.com   . my hardware part is done and is working but in group of 4+4+2. the attached here is the i am using. can you make desired changes in this case to make it run 10 matrices. if not 10 than at least 8.
[include.docx](https://github.com/wayoda/LedControl/files/1708154/include.docx)

DigaFabian, 2019-07-12:
@klarson18 It is hardcoded in the [LedControl.cpp](https://github.com/wayoda/LedControl/blob/master/src/LedControl.cpp#L50) file.

To get more than 8 matrices working go to your Arduino library folder, in my case it is located at:

> C:\Users\{Username}\Documents\Arduino\libraries\LedControl\src

There open the LedControl.cpp file with an editor like notepad++ and remove lines 50 and 51.
This way I was able to chain 15 matrices, though I am not sure what the maximum amount of matrices is.