wayoda/LedControl #19: Easiest way to display a number?

TommySharpNZ, 2016-09-07:
Hey guys, if I have an integer something like 62581.... Is there an easy way to get it displayed on an 8 digit 7 segment setup?

Do I have to split it up into the individual numbers or is there some trick I can do? I've seen some code snippets that divide by stuff and break it into 8 digits but I'm not sure how I'd ignore leading zeros and not end up with displaying 00062581....

Any help would be appreciated.

Tommy

funguypro, 2016-09-11:
I'm having the same problem - NEED HELP

wayoda, 2016-09-11:
Start from here:
http://wayoda.github.io/LedControl/pages/software.html#7Seg

closed 2016-09-11 by wayoda

funguypro, 2016-09-11:
Thanks! I also found some help here http://stackoverflow.com/questions/4261589/howto-split-a-int-into-its-digits