wayoda/LedControl #27: Still maintened ?

harlock974, 2018-07-09:
Hello Wayoda
Thanks for the great job. Ledcontrol library is incredibly useful (and I do love LED displays).
Is the library still maintened ?
I notice that interesting commits didn't received an answer since a long time.
All the best.

gordoste, 2019-05-03:
Not really. I have created a fork with what I think is the best of people's contributions.
* ESP8266 support
* Ability to choose between hardware and software SPI

Ryder17z, 2019-12-30:
> Not really. I have created a fork with what I think is the best of people's contributions.
> 
> * ESP8266 support
> * Ability to choose between hardware and software SPI

I would like to see a function to get the state of a segment/led added.

gordoste, 2019-12-30:
If you submit a patch to my fork I will happily accept it.