wayoda/LedControl #46: is there a way to know the state of a led

enfantme, 2022-02-27:
Hello ,
is there a way to know the state of a led
thank you in advance

CodeAsm, 2022-03-21:
No.

I wrote no, after writing the next bit that might be confusing and maybe even wrong. But the datasheet seems to indicate the internal SRAM (8x8) is write only for us. 

Besides reading DOUT pin (pin 24 on the MAX7219) I dont think you can read its old state before sleep or poweroff.
Thus, reading DOUT is a bit weird when your also writing it to the IC at the time you clock the data in.

If for instance, at powerloss you would like to resume what you where doing, instead write a state or message to onboard MCU eeprom/external and read from there. then alter your code to always clear the display upon startup and fetch the mode/state from nonvolatile mem. 

The datasheet shows a dualport sram but seems to be write only from "our" side. You cant seem to read from it once the data moved in. (maybe you can CLK cycle a bit to read the old data if its still there, but I doubt it)https://datasheets.maximintegrated.com/en/ds/MAX7219-MAX7221.pdf

Not to be used for data storage. most Arduino have onchip 1Kb of eeprom storage. external eeproms can also be used to store some message or status if required. but I would really think hard before using this.

wobisheng, 2022-04-14:
I think no
So I create a global array to store the state of leds